whatnick/ATM90E26_Arduino #2: Could you please explain what it means by semi working code using UART

waqas-ogc, 2016-07-28:
Could you please explain what it means by semi working code using UART , I mean  which features are working and which are not ?

whatnick, 2016-07-29:
Hi,

The code is not working on all platforms. Particulary the ESP8266 which was
the primary reason for the UART approach. The ESP8266 performs a WDT reset
after a few reads. If you are able to assist me in this regard you can
score an ATM90E26 breakout :)

Regards,

Tisham.

waqas-ogc, 2016-07-29:
Hi, 
Sure , I have ordered my ATM90E26 breakout and will arrive in a week or so then I would start work on both SPI and UART interfaces. I'm also working on ESP8266 based module WROOM-2. 
What about your SPI based code , is it working with ESP8266 ? 

Regards
Waqas

whatnick, 2016-07-29:
Hi,

The SPI code in arduino does not work on esp8266 due to lack of SPI mode 3
on esp8266. Not sure if it will work with another ESP compiler tool chain.

I am tracking the SPI mode3 issue. In order to make a wifo energy monitor
reliably with this chip you will have to use TI or Atmel alternative wifi
chips.

Regards,

Tisham.

waqas-ogc, 2016-07-29:
I also heard about this mode 3 issue. What if we use anther small Atmel /PIC MCU  in between ESP8266 and ATM90E26. The small MCU will read data from ATM90E26 over SPI and would send data to ESP8266 over UART ?

whatnick, 2016-07-29:
The UART mode should work with esp8266 once the WDT reset is fixed.

closed 2016-10-26 by whatnick