whatnick/ATM90E26_Arduino #3: how to read reactive energy?

lucasmaziero, 2016-07-28:
I am reading but not this as the correct value, I am also not consegindo calibrate the potecia factor, apparent and active power then with the correct valres
to calibrate these values ??

whatnick, 2016-07-29:
Sorry the sentence is a bit incoherent. What exactly do you mean ? What
values are you getting ? Are you using my breakout board or your own design
?

lucasmaziero, 2016-07-29:
I'm using the card "Energy Metering Demo board of Atmel"

Ugain (0x31), IgainL (0x32) is correct calibrated, I'm reading the correct values of Urms (0x49),  Irms(0x48).

I'm reading the Qmean (0x4B) with the function I created.

`double EnergyATM90E26::getRectivePower()
{
  short int qpower = (short int)commEnergyIC(1, Qmean, 0xFFFF); 
  return ((double)qpower);
}`

I do not know if I'm reading corrrtamente or has some error in my function, I thought the problem was in LPHI (0x24) to be calibrated, but when sending new values

`commEnergyIC(0, Lgain, 0x1D39); `

It is not being changed reitura the powerf (0x4D).

whatnick, 2016-07-29:
Gain on the channels should not change power factor. The atmel board uses a
shunt for sensing current so no phase errors due to CT etc. should be
introduced. I will have to look at you full code and run it for myself to
tell.Is this possible or are you using my sample as is ?

lucasmaziero, 2016-07-30:
Yes, I am using the original code only added its functions to a class, I'm using a 0.015Ohm shunt, from what I read in the datasheet was so calibrate Ugain (0x31), IgainL (0x32) that the reactive part already be calibrated.

I added this function to your code with correct result
```ruby
double getApparentPower()
{
  short int appower = (short int)commEnergyIC(1, Smean, 0xFFFF);
  return ((double)appower);
}
```
is already returns incorrect values ??

```ruby
double getRectivePower()
{
  short int qpower = (short int)commEnergyIC(1, Qmean, 0xFFFF); 
  return ((double)qpower);
}

double getAngle()
{
  unsigned short angle = commEnergyIC(1, Pangle, 0xFFFF); 
  return ((double)angle);
}
```
You can add to your code and test ?? it is possible to have errors in my functions.

thank you.

whatnick, 2016-07-30:
Please fork the library and make code changes there. Easier to work with it
that way. Patches via email are so last century. What sort of test load are
you using ?

lucasmaziero, 2016-07-31:
You can check in your project is correctly reading the values of the "power factor" and "reactive power"?

whatnick, 2016-10-26:
I have tested power factor with purely resistive Lamp and reactive (Compact Fluroscent) lamp. Both work as expected.

closed 2016-10-26 by whatnick