whatnick/ATM90E26_Arduino #4: Compilation issue in SPI example

lukeweston, 2017-05-08:
Hi Tisham, I'm getting some issues with the ATM90E26_SPI constructor, particularly around the CS parameter I think. ATM90E26_SPI(int) uses the default argument for parameter 1 which is not yet defined, etc etc. Compilation fails, using your unmodified SPI example, Arduino IDE 1.6.7. Any thoughts or suggestions?

whatnick, 2017-05-08:
Hi,

I have the sample code in the examples which instantiates 2 SPI ATM90E26 chips. This has been working okay with ESP8266 so far. Can you create your class following this template ?

Thanks for trying out the code. Any patches welcome.

Regards,

Tisham.

whatnick, 2017-05-09:
I have assigned a default SS pin. See if this works for you. Again I might need to set a pre-processor #define in a more generic way to work for most Arduino compatible boards. Suggestions for this are welcome.

per1234, 2018-02-15:
An `#if`...`#elif`...`#else` structure would be cleaner and easier to maintain:
```c++
#if defined(ARDUINO_ESP8266_WEMOS_D1MINI)  // WeMos mini and D1 R2
const int energy_CS = D8; // WEMOS SS pin
#elif defined(ARDUINO_ESP8266_ESP12)  // Adafruit Huzzah
const int energy_CS = 15; // HUZZAH SS pins ( 0 or 15)
#elif defined(ARDUINO_ARCH_SAMD) //M0 board
const int energy_CS = 10; // M0 SS pin
#elif defined(__AVR_ATmega32U4__) //32u4 board
const int energy_CS = 10; // 32u4 SS pin
#else
const int energy_CS = SS; // Use default SS pin for unknown Arduino
#endif  // defined(ARDUINO_ESP8266_WEMOS_D1MINI)
```

If the SS variable was not defined for the selected board you will get an unnecessary compilation failure (unnecessary since without this automated CS setting code the user could simply set it via the constructor). Here's an alternative code that solves that problem:
```c++
#if defined(ARDUINO_ESP8266_WEMOS_D1MINI)  // WeMos mini and D1 R2
#define ENERGY_CS D8 // WEMOS SS pin
#elif defined(ARDUINO_ESP8266_ESP12)  // Adafruit Huzzah
#define ENERGY_CS 15 // HUZZAH SS pins ( 0 or 15)
#elif defined(ARDUINO_ARCH_SAMD) //M0 board
#define ENERGY_CS 10 // M0 SS pin
#elif defined(__AVR_ATmega32U4__) //32u4 board
#define ENERGY_CS 10 // 32u4 SS pin
#else
#ifdef PIN_SPI_SS
#define ENERGY_CS PIN_SPI_SS // Use default SS pin for unknown Arduino
#else  // PIN_SPI_SS
#pragma message "SS pin could not be automatically determined for your microcontroller. You must set the CS pin via the constructor."
#endif  // PIN_SPI_SS
#endif  // defined(ARDUINO_ESP8266_WEMOS_D1MINI)
```
...
```c++
#ifdef ENERGY_CS
    ATM90E26_SPI(int pin=ENERGY_CS);
#else  // ENERGY_CS
    ATM90E26_SPI(int pin);
#endif  // ENERGY_CS
```
That will prevent the automatic CS setting code from ever causing an unnecessary error but unfortunately some variants don't define `PIN_SPI_SS` (The primary Arduino AVR Boards and Arduino SAMD Boards variants do). So the tradeoff for avoiding unnecessary failures is that the code will not automatically set the CS pin in some cases where assuming SS was defined would have allowed it.

whatnick, 2018-02-15:
Thanks so much for cleaning this up. Would you be able to formulate this
into a pull request. Otherwise I will look at this next week.

Regards,

Tisham.

per1234, 2018-02-15:
I'd be happy to submit a PR. Which code version do you prefer?

whatnick, 2018-02-15:
Please use the master branch. In future revisions I will looks at merging
SPI and UART code using a class hierarchy, since only the comms protocol is
different and there is lots of duplication.

per1234, 2018-02-15:
I have done so. Regarding my alternative proposal for code that would avoid unnecessary failures for boards without an SS variable defined, further research indicated that the `PIN_SPI_SS` macro was not used as universally as I might have hoped whereas `SS` was defined in almost all the common variants I looked at so I don't know whether it would be worth using that code. If that sort of issue was reported you could always just add another hardcoded assignment to the ones you already have.

whatnick, 2018-09-18:
If everything is working I would like to close this.

per1234, 2018-09-19:
You never merged my [pull request](https://github.com/whatnick/ATM90E26_Arduino/pull/7) so the problem still remains.

whatnick, 2018-09-28:
8-months and I never merged it "LGTM".

closed 2018-09-28 by whatnick