whatnick/ATM90E26_Arduino #5: Energy calculations and CF pulse output problem on atm90e26

dishak, 2017-06-13:
I have calibrated and getting all parameter values exact except enrgy and CF pulses.
The sys status(00h) is 00, 30h register is:C803(flexible mode) and 'N' line metering is enabled.
And it's CT based current measurement.
PL_Constant though inaccurate should at least o/p some increasing energy but, all values are zero.
To check energy accumulation I made PL_ConstantH=0 PL_ConstantL=0. But, now energy value in some constant number.
How do I resolve this?
The register initial settings  are as follows:
CommEnergyIC(0,SoftReset,0x78,0x9A,0x12); //Perform soft reset
  CommEnergyIC(0,FuncEn,0x00, 0x0C, 0x0E);  //enabled reverese energy flow
  //CommEnergyIC(0,SagTh,0x1F,0x2F,0x51); //Voltage sag threshhold
   CommEnergyIC(0,SagTh,0x10,0xBC,0xCF); 
  CommEnergyIC(0,CalStart,0x56,0x78,0xEE); //Metering calibration startup command. Register 21 to 2B need to be set
 CommEnergyIC(0,PLconstH,0x00,0xF8,0x19); //PL Constant MSB
 CommEnergyIC(0,PLconstL,0x8D,0x7F,0x2E); //PL Constant LSB
  //CommEnergyIC(0,PLconstH,0x00,0x00,PLconstH); //PL Constant MSB
  //CommEnergyIC(0,PLconstL,0x00,0x00,PLconstL);
  CommEnergyIC(0,Ngain,0x1D,0x39,0x7B);   //Line calibration gain
  CommEnergyIC(0,Lphi,0x00,0x00,0x24); //Line calibration angle
  CommEnergyIC(0,PStartTh,0x00,0x00,0x27); //Active Startup Power Threshold
  CommEnergyIC(0,PNolTh,0x00,0x00,0x28); //Active No-Load Power Threshold
  CommEnergyIC(0,QStartTh,0x00,0x00,QStartTh); //Reactive Startup Power Threshold
  CommEnergyIC(0,QNolTh,0x00,0x00,0x2A); //Reactive No-Load Power Threshold
  CommEnergyIC(0,MMode,0x90,0x22,0xDD); //Metering Mode Configuration. All defaults. See pg 31 of 
 datasheet.
  CommEnergyIC(1,CSOne,0x00,0x00,0x2C); //Checksum 1. Needs to be calculated based off the above values.
CommEnergyIC(0,SmallPMod,0xA9,0x80,0x2D);
  //Set measurement calibration values
  CommEnergyIC(0,AdjStart,0x56,0x78,0xFE); //Measurement calibration startup command
  CommEnergyIC(0,IgainL,0x2F,0x6E,0xCF);   //L line current gain
  CommEnergyIC(0,Uoffset,0x00,0X00,0X34);  //Voltage offset
  CommEnergyIC(0,IoffsetL,0x00,0X00,0X35); //L line current offset
  CommEnergyIC(0,PoffsetN,0x00,0X22,0X5B); //N line active power offset  //22  //4a
  CommEnergyIC(0,QoffsetN,0x00,0X01,0X3B); //N line reactive power offset  //01  //7b
   //CommEnergyIC(0,PoffsetN,0x00,0X00,0X39); //N line active power offset  //22  //4a
 // CommEnergyIC(0,QoffsetN,0x00,0X00,0X3A);
   CommEnergyIC(0,Ugain,0x55,0XF2,0X78);
  CommEnergyIC(0,IgainN,0x32,0X6B,0XD0);
  CommEnergyIC(1,CSTwo,0x00,0X00,0x3B);    //Checksum 2. Needs to be calculated based off the above values.*/
   //CommEnergyIC(0,CalStart,0x87,0x65,0x0C); //Checks correctness of 21-2B registers and starts normal //metering if ok
  //CommEnergyIC(0,AdjStart,0x87,0x65,0x1C); //Checks correctness of 31-3A registers and starts normal //measurement  if ok
  
  if (systemstatus&0xC000){
    //checksum 1 error
  }
  if (systemstatus&0x3000){
    //checksum 2 error
  }

}

Here calstart register I am not writing 0x8765 which checks correctness of 21-2B register and similarly I have commented adjstart which checks correctness for 31-3b register because, uncommenting these gave me interrupt  implies that the registers data are not correct and would stop the metering mode.
Can some one solve the issue? If I was not writing proper to the register then, I would not have got correct parameter values as well..

whatnick, 2017-06-13:
Here are the calculations, one of my clients is using now
 

 

MC = 1000 (imp/h)

Un = 230 (V)

Ib = 15 (A)

Gl = 1

Vl = Un / 940 = 245 (mV) (940 is the voltage divider coefficient)

Vu = Ib * 0.003 = 15 (mV) (0.003 is the shunt resistor value)

 

PL_Constant=

2677215.319

=

0028D9DF

PLconstH(0x21)=

0x0028

PLconstL(0x22)=

0xD9DF

 To get the absolute energy, the value is read from the following register.

APenergy

Forward Active Energy

Address: 40H

dishak, 2017-06-13:
Thank you sir,
I am reading from all energy registers but, damn thing I am not getting CF pulses or energy all register values read are zero.
`double GetImportEnergy(){
  //Register is cleared after reading
  unsigned short ienergy=CommEnergyIC(1,APenergy,0xFF,0xFF,0X3F);
  return (double)ienergy/10/1000; //returns kWh if PL constant set to 1000imp/kWh
}

double GetExportEnergy(){
  //Register is cleared after reading
  unsigned short eenergy=CommEnergyIC(1,ANenergy,0xFF,0xFF,0x40);
  return (double)eenergy/10/1000; //returns kWh if PL constant set to 1000imp/kWh
}
double GetAbsActiveEnergy(){
  //Register is cleared after reading
  unsigned short eenergy=CommEnergyIC(1, ATenergy,0xFF,0xFF, 0x41);
  return (double)eenergy/10/1000; //returns kWh if PL constant set to 1000imp/kWh
}
double  GetAbsReactiveEnergy(){
  //Register is cleared after reading
  unsigned short eenergy=CommEnergyIC(1, Rtenergy,0xFF,0xFF, 0x44);
  return (double)eenergy/10/1000; //returns kWh if PL constant set to 1000imp/kWh
}
double  GetReactivefwdEnergy(){
  //Register is cleared after reading
  unsigned short eenergy=CommEnergyIC(1, RPenergy,0xFF,0xFF, 0x42);
  return (double)eenergy/10/1000; //returns kWh if PL constant set to 1000imp/kWh
}`

whatnick, 2017-06-13:
Can you get Irms and Vrms values as non-zero. C801 indicates that
accumulator is not running. Do you have an Atmel representative to contact ?

dishak, 2017-06-13:
No I don't have atmel contacts.
I checked the Enstatus (46H)=03. Sys status (00H)=00.
Are my checksomes correct?

whatnick, 2017-06-13:
If the error/warn is not raised checksums are correct. You may also be
reading accumulator too fast. Wait for status to be 2801 indicating
accumulator is populated.

dishak, 2017-06-13:
ya I gave some delays.Now the values I get are: meter status(0X46h)=2003 (in your case 2801 because line measurement mode but, in my case it is 'N'  line measurement mode) so the value is:2003 now..
But, I have a doubt i.e:2003 indicates as per bit (page 43) that reactive power is reversed (in your case and my case also) is that fine?
But, still no energy accumulation.

dishak, 2017-06-13:
For my code I am writing the self calculated CRC byte.
here is the code which only performs read/write operations:
unsigned short CommEnergyIC(unsigned char RW,byte address, byte msb_byte,  byte lsb_byte, byte crc_byte){
  address|=RW<<7;   
  unsigned short output;
  
  byte fn_write[5]={0xFE,address,msb_byte,lsb_byte,crc_byte};
  byte fn_read[3]={0xFE,address,address};
  byte hexIn[3]={0};
  byte hexIn1[1]={0};
  int readbytes;
//  portTwo.listen();
  
  
  if(RW)
  {

  Serial.write(fn_read, sizeof(fn_read));
  int readbytes=Serial.readBytes(hexIn,3);
  }
  else
  {
  Serial.write(fn_write, sizeof(fn_write));
  readbytes=Serial.readBytes(hexIn1,1);
}
 delay(100);
 output=hexIn[0]*256+hexIn[1];
 return output;
}
but,
then, Actually I used your code where we just need to send the address and bytes to be sent there is crc calculator and meter status got  was (0x46h register):2003h and sys status :00h.

dishak, 2017-06-13:
now I am getting CRC values: checksome1,checksome2:
here: checksome1-->21 to 2bh  is: 21CA8.
And all other high byte ,lowbyte from 21H to 2BH is: 0000,0000,0000,0000,1D39,0000,0000,0000,0000,0000,9022. resply checksome 1 I got from meter: 138408.
And checksome2--->31 to 3Bh is:
5678,5572,2f6e,326b,0000,0000,0000,0000,0000,0022,0001 resply
and checksome2 I got from meter: 240100.
Actually how do I calculate checksome?they have mentioned a procedure in datasheet and I am getting some decimal values..which is not correct can u help me and explain the calculation and verify if values what I have got is correct checksome or checksome error.

whatnick, 2017-06-13:
For testing the chip I just read back and rewrote the checksum assuming all
comms were fine and I was getting the same checksum over multiple
configuration runs. Looks like you are making progress, add a bit more
delay and read the accumulator.

dishak, 2017-06-13:
I did give 1second delay after write command to the energy IC but, energy accumulation is very random and not constantly increasing.
In datasheet they have mentioned that the energy registers are read and clear that means every time we read it's going to clear then, how will the accumulation happen? Energy registers should not clear after every read if they do so then, in program how can I consider the read energy as accumulated?

whatnick, 2017-06-13:
Yes the energy registers are meant to be read slowly and cleared once read
to prevent overflow.

dishak, 2017-06-14:
ok, Now I will give long delays and see once but, in PL_constant measurement the formula has basic current Ib what is it? Is it the primary current of CT/minimal current in primary I don't know exactly.

whatnick, 2017-06-20:
Do you clearly understand the principal of operation of the metering IC ? The IC only takes limited voltages into its ADC. Everything else is scaled using the registers to reflect real values. The multiplier should be obtained using the primary current.

closed 2017-06-20 by whatnick

XingyumLee, 2019-05-17:
> Here are the calculations, one of my clients is using now
>
> MC = 1000 (imp/h)
>
> Un = 230 (V)
>
> Ib = 15 (A)
>
> Gl = 1
>
> Vl = Un / 940 = 245 (mV) (940 is the voltage divider coefficient)
>
> Vu = Ib * 0.003 = [...]

Have he created such circuit to test? Can't I do it to calculate?

joaovsmathias, 2019-05-20:
> Then,once I read the energy register I should be adding the energy read previously and energy read instantly no?
> Now I will give long delays and see once but, in PL_constant measurement the formula has basic current Ib what is it? Is it the primary current of CT/minimal current in primary I don't know exactly.

Hi dishak,

You already resolved, or find out something new about the problem?
 I'm going through the same difficulties at the moment.

XingyumLee, 2019-05-27:
how do you measure the cf frequency?whats the relationship between frequency and energy/

papay1, 2023-05-01:
I am facing the same problem with m90e36a , getting non zero voltage and current value but energy register is zero and i also add sufficient delay to accumulate but still getting error .