whatnick/ATM90E26_Arduino #10: GetImportEnergy just returns zero and documentation

agaelema, 2018-07-18:
Hi, I'm using the ATMEL M90E26 breakout with NodeMCU. Everything is working well except the function "GetImportEnergy()".

The function always returns zero. I tried to catch the RAW value of the register (creating a new function "GetImportEnergyRaw()") to verify the returned value, but remains zero.

In the code I read the values every 10 seconds, the result is shown below:

_**Sys Status:C002
Meter Status:C801
Voltage:118.16
Current:1.354
Active power:158.00
p.f.:0.99
kWh:0.0000
kWh RAW:0
freq:60.03**_

I tried to increase the interval between reading, but the value returned remains 0, even using an interval of 10 minutes.

Whats the problem?

Is difficult to understand the internal code due to the lack of documentation about the constants used in the calculus to get the hex values to write on the device, e.g., the line voltage, the current, etc.

Thanks.

agaelema, 2018-07-19:
Hi,

I managed to solve the problem, at least partially.

The problem is that I was compensating the phase shift but was not recalculating the checksum. After recalculating it worked fine.

To contribute to the project I created two new functions to get the raw value of the accumulators besides adding some #defines. I also updated the excel sheet to calculate both checksums.

The current problem is to understand why the accumulator returns half of the energy consumed, for example, a load of 180W is registering 0.09kWh. Is this error related to any gain or recorder?

Thanks

whatnick, 2018-09-18:
Perhaps there are multipliers in the Arduino code not present in the registers. Then you need to apply the Arduino multipliers to data read from the energy register as well.

muratgortekin, 2019-04-29:
Hi, I applied the changes you specified, but still returns the value to 0. Is there any resolution on this? What else am I doing wrong.
@agaelema @whatnick @andriyadi