whatnick/ATM90E26_Arduino #13: ESP32 UART read failed / write failed

Joaovma, 2018-09-27:
Hi,
I tried to use this code for UART comunication with ATM90E26 but without success.
I keep receiving "Write failed" on the begining and then i get a lot of "read failed".
I tried to change the pins but without success as well.

Could you help me ?

I'm using a ESP32. 
I tried this same code with a ESP8266 and it worked! I didnt get the "R/W failed".
But changing to ESP32, it does not work, because I get the "R/W failed" as I mentioned above.

Any suggestions ?
Thank you.

whatnick, 2018-09-28:
Your issue does not mention which ESP32 pins you used and what code you ran
exactly. Moving from one controller to another and hoping it will work is
rarely a good plan. Please post a code snippet and wiring diagram with
ESP32.

Joaovma, 2018-10-02:
Hello Whatnick,
I used the exactly same code with the same pins that was presented at your tutorial(GPIO16 and GPIO17 - RX2 and TX2) and the code at this repository for UART in order to communicate with ATM90E26 breakout board. I also tried another pins such as 1 and 3(RX0 and TX0) but i got the same result.
In addition, I am testing the uart communication between two ESP32 and works well. However, when we test UART communication(which use Hardware serial library) with the ATM microcontroller it does not work. 
In your tutorial, it was presented ESP8266 but recently someone uploaded the repository with the UART code for ESP32.  When we get that code and test it with ESP8266 and ATM90E26 it worked, but not between ESP32 and ATM90E26.
Again, we used the exactly same code presented on this repository , but I attached a ZIP file with the code.
Could you help me by sending me an example of any ESP32 UART code communicating with ATM90E26 Breakout Board? It will be very helpfull.
[ATM90E26_UART.zip](https://github.com/whatnick/ATM90E26_Arduino/files/2439305/ATM90E26_UART.zip)

whatnick, 2018-10-03:
Please fork my repository and push code to it for easier review. ESP32 UART
mode is tested to work.

francescolavra, 2018-11-01:
The example code for ESP32 uses pins 19 and 23 (RX and TX, respectively), not pins 16 and 17. Try connecting those pins to the ATM90E26 board and it should work.

Joaovma, 2018-11-01:
> The example code for ESP32 uses pins 19 and 23 (RX and TX, respectively), not pins 16 and 17. Try connecting those pins to the ATM90E26 board and it should work.

I tried pins 19 and 23 already. But I always get this same result (read failed / write failed) (image attached)
It is the same code provided here on this repository.
I attached the code on my last comment above.

Any other suggestions ?

[image]

lucascoelho91, 2018-11-04:
Hi @Joaovma, @whatnick, @francescolavra 
I'm also facing the same problem @Joaovma experienced. Using the example code available [here](https://github.com/whatnick/ATM90E26_Arduino/blob/master/examples/ATM90E26_UART/ATM90E26_UART.ino), it worked fine for ESP8266 but not for ESP32. I'm getting the same write failed problems @Joaovma reported.

Here is how I plugged the ESP32 and the ATM90E26:
[image]
I've tried also inverting the TX/RX connectors but it did not work. Also tried to use other ports, but no success. I suspect it should have something to do with the ESP32 board model, but I have no idea what else I could try. Any ideas will be highly appreciated.

jdeglavina, 2018-11-04:
FYI, I have a fork of whatnicks code working with the esp32 using SPI.
Although, I updated the library to use the ATM90E32. The SPI part of the
code should be the same though.

whatnick, 2018-11-07:
> @jdeglavina <https://github.com/jdeglavina> thanks, your library worked
> fine! And it also does not have the sleep(20) command.
> But I am still confused about why the original library did not work for me.

Please send me a PR from your fork to review the diff and comment.

XingyumLee, 2019-05-23:
i use the esp8266 and your library,why on the same circuit at first theres outcomes,but serveral times after,theres only 65535 outcomes why

jamescarlson, 2020-09-28:
Not sure if this is related to the issues here, but upon upgrading from the ESP8266 arduino libraries from 2.4.2 to latest (2.7.x), I was not able to get the code working again. After adding several print statements in the main `CommEnergyIC` function, it looked like the problem I was experiencing was one of having information left in the Serial buffer from the last call to `CommEnergyIC`, which would get picked up on the next call, and thus mix up the outputs.

I solved this mixup by adding a read call on [line 42 of energyic_UART.cpp](https://github.com/whatnick/ATM90E26_Arduino/blob/master/energyic_UART.cpp#L42) to clear out any data that's leftover from last call. Kind of weird that an upgrade to the libraries (SoftwareSerial included) would cause this but it's got my power monitor working and with the new libraries, not crashing every 15 minutes. 

If this also works for anyone else, please let me know.

whatnick, 2020-09-28:
This is great. Would love to have this in a PR. Testing all versions of
libraries against real hardware is a bit impossible. I am looking at
setting up some CI with GitHub + self hosted runners.