whatnick/ATM90E26_Arduino #14: SoftSPI support for this library

Jimyxy, 2018-11-26:
Hello Tisham
I saw the great work you have done in the energy metering with the ATM90E26. Thank you very much for this library it is very useful for non code experts like me. I have this request, Is it possible you can implement support for softSPI use?. There are some softSPI examples for other chips done with [DigitalIO  library](https://github.com/greiman/DigitalIO) like the [RFID via SoftSPI](https://github.com/abhaysbharadwaj/RFID-RC522), I thought it clould be easy, I tried to do it by my self (I was Wrong).  _I have tried to connect the ATM90E26 via softSPI with no success._ Thanks in advance.

whatnick, 2018-11-29:
Would you be able to supply details of the board you tried. SPI is very
time sensitive and softspi may fail if timing is not guaranteed.

Jimyxy, 2018-11-29:
> Would you be able to supply details of the board you tried. SPI is very time sensitive and softspi may fail if timing is not guaranteed.
> […](#)

The SPI connections the board is now is as follow:

ATM90E26 ---   --- Arduino promini - ATmega328p
MOSI 27    ---   --- D5
MISO 26    ---   --- D6
CLK   25    ---   --- D7
CS     24    ---   --- D8

Now I was able to pick up the pin 12 (USEL) in the ATM90E26 and pull this up to VCC, then it is selected as UART communication mode, I making some tests to achieve the readings using software serial library with the arduino. It is not perfect (the wire over the IC is ugly) but does the job. Anyway I think this is a great library, I think it is good that it has soft SPI Support and I can remove the ugly wire. Thanks.