whatnick/ATM90E26_Arduino #17: Calibration Problem

muratgortekin, 2019-04-24:
Hello there,

I have a problem with calibration. I have used the following link in the .ino and excel calibration file.
https://github.com/muratgortekin/atm90e26_esp8266
I'm reading the values by zero. I'm using a current transformer. I can't change what I did. What's wrong with that?

whatnick, 2019-04-24:
Have you changed the checksum value to match your new calibration
parameters? One of easiest approaches is to apply a multiplier in your
Arduino firmware.

muratgortekin, 2019-04-24:
How do I read Checksum? Can I see directly from the serial monitor? Checksum2 is giving an error. Do I have to write the value? Does this ATM90E26 chip calculate myself? Can you tell me the steps, please?

whatnick, 2019-04-24:
Please look at the other issues here and the code. Yes the ATM90E26
calculates it, if you read the docs please implement the same in Arduino
code ..

muratgortekin, 2019-04-25:
Thank you for your help. I think the calibration job looks okay for the voltage. But I can't read the current value. It always sends the value 0.

muratgortekin, 2019-04-25:
Are the MMD0-MMD1 pins connected to GND or connected to VDD? Is it the problem? I have implemented the GND for MMD0 and the VDD for MMD1.

XingyumLee, 2019-05-23:
do you know how to work out the Lgain,what do the  two frequency  mean?