whatnick/ATM90E26_Arduino #24: Unable to communicate with ATM90E26

maheshbhome, 2020-09-24:
Hi All,
         I am working with ATM90E26 metering IC. I have implemented circuit as per Application note,  use Arduino uno board   and used source code from whatnick/ATM90E26_Arduino but getting 00 reply for every read as per source code, please help me out on same.

whatnick, 2020-09-24:
Atm90e26 is a 3.3v device. I believe UNO is 5v. Need more details on your
exact circuit and SPI vs UART mode. Also please use a logic analyzer to
check signals on you wires.

maheshbhome, 2020-09-25:
Thanks for your reply, Now I have connected ATM90E26 to Beaglebone black wireless using SPI. 
 I have some queries, 
1) ATM90E26 need to be calibrated or it can be run without calibration.
2) if Calibration required, then  is it  ok that values set as per excel sheet provided by you in arduino code.
3 ) [I](url) am able to read voltage  and frequency but every time I get new values.

[image]

maheshbhome, 2020-09-25:
one more query , is shunt resistance is necessary in circuit.

maheshbhome, 2020-10-29:
hi, I am unable to read APenergy register kindly help me out on same.

closed 2020-10-29 by maheshbhome

Yogesh-2897, 2021-07-16:
Hii, I have implemented the circuit as mentioned in the application note for ATM90e26, But i am not able to communicate with the IC through spi. Please help