whatnick/ATM90E26_Arduino #27: ATM90E26 FeatherWing (Huzzah) crashing

MrPivato, 2020-10-08:
Hi! I am using an `ATM90E26 Featherwing v1.5`, with an `Adafruit Huzzah`.
The problem comes when I try the example skecth `ATM90E26_Full_App.ino`, I only updated it to run newer versions of `ArduinoJson` and `WifiManager`, as you can you can see here: [My Fork](https://github.com/MrPivato/ATM90E26_Arduino).

When it tries to read the values from the meter, the result is a crash, but connection to WiFi and Thingspeak is ok. 
Here is the error message
```
{"auth":"XXXXXXXXXXX","server":"api.thingspeak.com"}connected...yeey :)
Key:XXXXXXXXXXXX
Server:api.thingspeak.com
Start ATM90Checksum 1:FFFF
Checksum 2:FFFF
Checksum 1 Error!!
Checksum 2 Error!!
Checksum 1:FFFF
Checksum 2:FFFF
Checksum 1 Error!!
Checksum 2 Error!!

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```

If anyone could help me solve this problem I would be grateful, thanks in advance!

whatnick, 2020-10-10:
In firmware interaction of the various parts is crucial. I haven't checked the code in your fork, but do you have the SPI/ UART jumpers set correctly ? Can your monitor transactions on the serial bus using a logic analyser ? Some images of your stack up and jumper settings will help with this issue. Looks like no values are being read from the ASIC hence the FFFF.

MrPivato, 2020-10-16:
I am using the stacked version. Here is the image:
[image]

MrPivato, 2020-10-26:
Hello @whatnick 
Any tips?

whatnick, 2020-11-24:
Could I please see the jumper settings underneath.

MrPivato, 2020-11-26:
You mean here?

[image]

whatnick, 2020-11-26:
Please set jumpers for SPI or UART mode as appropriate. No jumpers have
been set. Here please check the PCB or code repo for the jumper options.