whatnick/ATM90E26_Arduino #29: Voltage Sampling Issue

sdk694, 2021-04-03:
Hi ,

I am using your code to interface ATM90-E26 with ESP8266. Ran the default code without any calibration and got this output, the output should be around 230V but I am getting 0.03V. The circuit that I have implemented is attached,  The jack gives an output of 12vac and Vp is measure across r2 which is fed to the ATM and  the ATM is powered externally with 3.3 V adapter. Any suggestions?

[image]

[image]
 with a  3.3V  adapter.

whatnick, 2021-04-03:
Please supply the full schematic or compare it to the [ATM90E26 Featherwing](https://github.com/whatnick/ATM90E26_Featherwing_KiCAD) for reference. It may also be a good idea to buy/make the featherwing design yourself for a complete reference before jumping into a full custom design.

sdk694, 2021-04-04:
I have used UART for communication, following are the connections that I have made 
USEL - S.C with 3.3 - (UART)
MMDO - S.C with 3.3
MMD1- S.C with GND
and have connected the CT to I1 and I1P . I have provided the schematic SS and the PCB's photograph for your reference.

[image]
[image]
[image]
[image]

whatnick, 2021-04-04:
Hi thanks for the additional info. The issue seems to be unpopulated R1/R2 resistors in the PCB , which means no voltage is getting to the energy monitor except whatever the FR4 resistance and capacitance coupling is getting to it. To verify please use a multi-meter to check the voltage at the VP test point. 

Populate the 1k / 100k voltage divider ladder and try again.

sdk694, 2021-04-04:
Thank you. I made the corrections that you suggested. Now I am getting the Voltage as 112.25  V. The voltage should be around 230V. Any suggestions?

whatnick, 2021-04-04:
Since some voltage is being sampled it is time to perform calibrations. Please see docs here : https://github.com/whatnick/ATM90E26_Arduino/tree/master/docs . TL;DR your options are to change your voltage divider ladder resistors or calibration registers to get the actual values.

sdk694, 2021-04-04:
I tried do change the Ugain value using the excel sheet that was present on the repository.  Once the changes were uploaded the voltage was coming out to be 0.

whatnick, 2021-04-04:
Did you read all of the docs and update the CRC values ? What do the status registers say. Please lodge some consulting time with me. Unfortunately this is the limit of open-source support I can offer. The mistakes with missing voltage divider ladder were easily avoided. Would you like to give me some idea of your technical background so that I can tune my support appropriate level of assumed knowledge.

PadmaIoT, 2023-07-10:
@sdk694 I am having the same problem too. Used 1k / 100k. SShowing voltage of 142V AC. The difference with you is I am using SPI

whatnick, 2023-07-10:
The code is indicative only for a sample setup I had at the time. Please
use calibration parameters to get the value you require. Chips may differ
in internal resistance and voltage divider ladder resistors and supplies
like transformers and even direct AC have their own internal resistance.

PadmaIoT, 2023-07-11:
I changed the Ugain according to the actual & measured voltage for calibration (from the excel sheet), but the voltage gets down more after changing the Ugain.