whatnick/ATM90E26_Arduino #31: STM32 blupill support

PadmaIoT, 2022-06-21:
Hi,
Is this library compatible with STM32 bluepill?
If compatible then how to wire?
I am very new in single-phase metering.

whatnick, 2022-10-10:
Hi, this has not been tested on the STM32 Bluepill. I do have some of them in stock and this is great incentive to test. The ASIC supports both UART and SPI modes so wiring should be as below.

[STM32 Pins](https://sites.google.com/view/makeriot2020/home/tutorials/stm32-blue-pill/programming-the-bluepill-with-arduino-ide)

PadmaIoT, 2022-10-10:
Hello @whatnick ,
This library is not compatible with STM32 blue pill. I can not compile it. Then I tried to look into your CPP & h file so that I could make the library work for STM32. But I failed there too. After failing I started to build my version of the library on my own but it was tough. the calculations resisters etc just blew me up . So I am totally counting on you.

whatnick, 2022-10-10:
Yes, I just installed support for the STM32 boards in my IDE and I have few
days free to make it work.

PadmaIoT, 2022-10-10:
Thank you, Mister. I will be really happy & obliged for your update.

whatnick, 2022-10-10:
Could you please order this board so that we can be on the same page regarding wiring and usage. https://www.pcbway.com/project/gifts_detail/ATM90E26_Breakout.html

PadmaIoT, 2022-10-11:
Sure do. I have a ques because I could not understand a thing from the datasheet and from your design too what is the function of MMD1 & MMD0 ?

whatnick, 2022-10-11:
The MMD0 and 1 jumpers set "Measurement Mode". There are 4 possible modes.
I am having trouble sourcing a Bluepill, I will see if I can get a
STM32F103 Nucleo.

whatnick, 2022-10-11:
> Sure do. I have some other ques because I could not understand the portion of the datasheet and your design+firmware too.
> 
> 1. what is the function of MMD1 & MMD0?
>> These set metering modes, ie. Live on, Neutral on and Tampering detection. Use the default live on and you should be fine.
> 2. Why there is two for PF, VRMS, CRMS, and Real power?
>> There is code for dual ASIC usage with CS (Chip select) based multiplex. So you can get 2 readings, useful for single-phase grid connected solar setups common in Australia where I live and this was developed.
> 3. What if I use ring CT instead of split CT?
>> The choice of CT is not material to the Arduino library as long as you have within spec proxy for the current with low waveform distortion, it can be a ring CT, split CT or shunt resistor. The usage of the library is not affected by it.

From my estimation you need some Electrical fundamentals, I will leave this as a warning since working with 240v (LV) requires some safety precautions. Please study these before further work to avoid injury to yourself or damaging equipment.

Please test code in PR here : https://github.com/whatnick/ATM90E26_Arduino/pull/32

PR #32 (2022-10-30): Support STM32F1xx Bluepill variant
Closes #31 
[x] Successful compile of UART Mode
[x] Add table diagram for UART mode
[x] Successful compile of SPI mode
[x] Add this platform to supported one in library metadata
closed 2022-10-30 by whatnick